realsenseai/RealSenseID #116: Corrupt JPEG data: 115936 extraneous bytes before marker

zeynepkoyun, 2021-08-05:
Hi @gabime 

I get the error "Corrupt JPEG data: 115936 extraneous bytes before marker" when running the viewer.py code in python and the image looks corrupted.

gabime, 2021-08-08:
Please provide more info about os and compiler and python versions.

zeynepkoyun, 2021-08-08:
> Lütfen işletim sistemi, derleyici ve python sürümleri hakkında daha fazla bilgi verin.

Hi @gabime ,
Ubuntu 16.04, python3.6 and the project version I am using is v0.23.0

gabime, 2021-08-08:
Please try the rsid-cli tool to see if the problem is python specific or from the env (press ‘c’ to get some preview frames). 

Also, please use ubuntu 18 or later.

zeynepkoyun, 2021-08-09:
Hi @gabime 
I get the error I mentioned, especially when the device starts to heat up, but when I get an error, when I look at the realsenseID application in windows, the images look normal.
As you said, let me check with "rsid-cli" in ubuntu and share the results with you.

zeynepkoyun, 2021-09-14:
Hi @mdisg 
Did this piece of code exactly solve your problem?
When I made this change and tested it, I noticed that the recorded images were wrong, but the display does not show the wrong image, but the camera is still broken.

mdisg, 2021-09-14:
@zeynepkoyun yes the wrong images are ignored/dropped e.g. by the client in my case Firefox or Chromium so the data is still corrupted as I mentioned in my last post.

zeynepkoyun, 2021-09-14:
@mdisg @gabime 
I am working with ubuntu 16.04. When I get a distorted image from the camera, the image looks fine when I look at it with the windows application.