realsenseai/RealSenseID #119: RUN TIME CRC error

ryan185, 2021-08-16:
I am using windows 10 with Python 3.8 and getting error in enroll.py and all other sample too however viewer.py is just showing live feed but it stops if press button

[2021-08-16 10:39:08.992] [debug] [WindowsSerial] Opening serial port COM3
User id to enroll: 12
[2021-08-16 10:39:12.742] [debug] [NonSecureSession] Start session
[2021-08-16 10:39:12.743] [debug] [PacketSender] Sending packet 'o'
[2021-08-16 10:39:12.747] [debug] [PacketSender] Waiting packet..
[2021-08-16 10:39:12.753] [error] [PacketSender] Got invalid crc. Expected: 561. Actual: 20100
[2021-08-16 10:39:12.754] [error] [NonSecureSession] Failed to recv device start session response
[2021-08-16 10:39:12.754] [error] [FaceAuthenticatorImpl] Session start failed with status 108
on_result EnrollStatus.CrcError
Traceback (most recent call last):
  File "enroll.py", line 27, in <module>
    f.enroll(user_id=user_id, on_hint=on_hint, on_progress=on_progress, on_faces=on_faces, on_result=on_result)
RuntimeError: CrcError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "enroll.py", line 31, in <module>
    print('Users: ', users)
TypeError: __exit__(): incompatible function arguments. The following argument types are supported:
    1. (self: rsid_py.FaceAuthenticator, arg0: capsule, arg1: capsule, arg2: capsule) -> None

Invoked with: <rsid_py.FaceAuthenticator object at 0x000002D0891FB2F0>, <class 'RuntimeError'>, RuntimeError('CrcError'), <traceback object at 0x000002D0892194C0>
[2021-08-16 10:39:12.761] [debug] [NonSecureSession] Close session
[2021-08-16 10:39:12.765] [debug] [WindowsSerial] Closing serial port

zeynepkoyun, 2021-08-23:
Hi @ryan185 

Can you check if similar error was resolved with software update.
https://support.intelrealsense.com/hc/en-us/community/posts/4404417377299/comments/4404467141651

ryan185, 2021-08-24:
this solution is for Linux i am working on Windows will it work?

zeynepkoyun, 2021-08-25:
Hi @ryan185 

CrcError Error is an error that is received when the firmware update is not done correctly, so I gave this resource as an example. Did you update the firmware correctly?
E.g; When updating from the command line with windows, can you check the output by giving the following command its own binary file?
rsid-fw-update.exe --file F450_x.x.x.xxxx._SKU1_SIGNED.bin --port COM8

RealSenseSupport, 2021-10-22:
@ryan185, Is this issue fixed, after upgrade the FW?