realsenseai/RealSenseID #122: Is It Possible to use F455 Directly with an Arduino?

korroktheslavemaster, 2021-08-28:
Hi
I just purchased a F455 sensor. From the documentation it looks like compiling the `RealSenseID` library is necessary to establish any kind of communication with the device. I was wondering if there is some other way of communicating simple instructions (trigger for auth on a GPIO, and receiving response on another GPIO/UART pins) so that I can directly interface my F455 with an arduino microcontroller and use it that way, instead of attaching a relatively expensive SBC like raspberry pi or jetson nano to it?

Also is there any documentation for using the GPIO pins exposed through the AUX port? I couldn't find much reference to them beyond the F455/F450 datasheet pdf.

Monica2511, 2021-08-30:
You can't connect directly with arduino, you need an device have OS  for compiling library like raspberry/ jetson nano

RealSenseSupport, 2021-09-28:
I think, you can try to use F450 module with redesigned interposer.
reference to "Figure 8-2. UART Interposer Reference Design" in the datasheet. download from https://www.intelrealsense.com/download/14251/

and than re-write this software.