realsenseai/RealSenseID #123: Viewer video lagging

omigus, 2021-08-30:
I am using jetson nano with Python3.6.9 and getting video lagging in viewer.py .How should I solve this problem?

Thank You

[image]

omigus, 2021-08-30:
UPDATE

i found error in logs  here

Version: 0.25.0
[2021-08-31 00:21:24.700] [debug] [Utilities] capture devices 1
[2021-08-31 00:21:24.728] [debug] [Preview] Preview started!
Corrupt JPEG data: 143693 extraneous bytes before marker 0xd1
Corrupt JPEG data: 857 extraneous bytes before marker 0xd2
Corrupt JPEG data: premature end of data segment
Corrupt JPEG data: premature end of data segment
Corrupt JPEG data: 190 extraneous bytes before marker 0xd2
Corrupt JPEG data: premature end of data segment
Corrupt JPEG data: premature end of data segment
Corrupt JPEG data: premature end of data segment
Corrupt JPEG data: 262 extraneous bytes before marker 0xd1
Corrupt JPEG data: 536 extraneous bytes before marker 0xd0
Corrupt JPEG data: 1086 extraneous bytes before marker 0xd1
Corrupt JPEG data: 269 extraneous bytes before marker 0xd2
Corrupt JPEG data: premature end of data segment
Corrupt JPEG data: 1630 extraneous bytes before marker 0xd2
Corrupt JPEG data: premature end of data segment
Corrupt JPEG data: 685 extraneous bytes before marker 0xd2
Corrupt JPEG data: 328 extraneous bytes before marker 0xd0
Corrupt JPEG data: premature end of data segment
Corrupt JPEG data: 118 extraneous bytes before marker 0xd1
Corrupt JPEG data: premature end of data segment
Corrupt JPEG data: premature end of data segment
Corrupt JPEG data: 1668 extraneous bytes before marker 0xd0
Corrupt JPEG data: premature end of data segment
Premature end of JPEG file
Corrupt JPEG data: premature end of data segment
Premature end of JPEG file
Corrupt JPEG data: 353 extraneous bytes before marker 0xd1
Corrupt JPEG data: 264 extraneous bytes before marker 0xd1
Corrupt JPEG data: premature end of data segment
Corrupt JPEG data: 210 extraneous bytes before marker 0xd1
Corrupt JPEG data: 121 extraneous bytes before marker 0xd1
Corrupt JPEG data: 47 extraneous bytes before marker 0xd2
Corrupt JPEG data: premature end of data segment
Corrupt JPEG data: premature end of data segment
Premature end of JPEG file
Corrupt JPEG data: premature end of data segment
Premature end of JPEG file
Corrupt JPEG data: premature end of data segment
Corrupt JPEG data: premature end of data segment
Corrupt JPEG data: 160 extraneous bytes before marker 0xd0
Corrupt JPEG data: 135391 extraneous bytes before marker 0xd2
Corrupt JPEG data: 924 extraneous bytes before marker 0xd1
Corrupt JPEG data: 206 extraneous bytes before marker 0xd0
Corrupt JPEG data: premature end of data segment
Corrupt JPEG data: 116 extraneous bytes before marker 0xd2
Corrupt JPEG data: premature end of data segment
Corrupt JPEG data: 1071 extraneous bytes before marker 0xd0
Corrupt JPEG data: 162 extraneous bytes before marker 0xd0
Corrupt JPEG data: premature end of data segment
Corrupt JPEG data: premature end of data segment
Corrupt JPEG data: 142 extraneous bytes before marker 0xd2
Corrupt JPEG data: 153 extraneous bytes before marker 0xd0
Corrupt JPEG data: 17440 extraneous bytes before marker 0xd9
Corrupt JPEG data: premature end of data segment
Premature end of JPEG file

panudet-24mb, 2021-08-30:
Same as mind , I also got those problem with jetson-nano 
@omigus have u try build those lib again ?
P.S. while I running sample program CPU ran so high 100% percent of my Jetson CPU was eaten .

mdisg, 2021-08-31:
@omigus you're not alone with that kind of errors see issue #116

panudet-24mb, 2021-09-07:
@mdisg  I try to implement your solution in #116 But still got problem with image shape resolution , any suggestion to figure it out .