realsenseai/RealSenseID #125: Return from Standby mode

mdisg, 2021-09-01:
Recently I have played around with the standby mode using the C# Wrapper Class `Authenticator`.
After activating the standby mode enabling the camera preview via `Preview.Start` returns `true` but no preview image is received. The callback will not be called in that case. I then tried to do an authentication right after the reconnect and after that the preview works so I think the camera is still in standby mode even after the reconnect to the device. The start result for the preview should be `false` in that case in my opinion. Could you also please add a way/method to wakeup the unit without starting a dummy authentication e.g. by adding a Method e.g. `WakeUp()`?

My initial approach was:

- Call `Connect()` on my `Authenticator` instance.
- Call `Standby()` and `Disconnect()`
- Wait few seconds
- Call `Connect()` again on the same authenticator instance.
- Wait few seconds
- Call `Start()` on a newly created `Preview` instance.

The code:
```
var serialConfig = new SerialConfig { port = "COM2" };
var authenticator = new Authenticator();
if (authenticator.Connect(serialConfig) == Status.Ok)
{
    authenticator.Standby();
    authenticator.Disconnect();
    Thread.Sleep(3000);
    if (authenticator.Connect(serialConfig) == Status.Ok)
    {
        Thread.Sleep(3000);
        var preview = new Preview(new PreviewConfig
                                      {
                                          cameraNumber = 0,
                                          previewMode = PreviewMode.MJPEG_720P,
                                          portraitMode = true
                                      });
        if (!preview.Start(PreviewImageReceived))
        {
            Console.WriteLine("Start preview failed");
        }
    }
}
```

The following code with call to `authenticator.Authenticate(new AuthArgs());` works:
```
var serialConfig = new SerialConfig { port = "COM2" };
var authenticator = new Authenticator();
if (authenticator.Connect(serialConfig) == Status.Ok)
{
    authenticator.Standby();
    authenticator.Disconnect();
    Thread.Sleep(3000);
    if (authenticator.Connect(serialConfig) == Status.Ok)
    {
        Thread.Sleep(3000);
        authenticator.Authenticate(new AuthArgs());
        var preview = new Preview(new PreviewConfig
                                      {
                                          cameraNumber = 0,
                                          previewMode = PreviewMode.MJPEG_720P,
                                          portraitMode = true
                                      });
        if (!preview.Start(PreviewImageReceived))
        {
            Console.WriteLine("Start preview failed");
        }
    }
}
```

RealSenseSupport, 2021-09-10:
Have you try cameraNumber with different value? I guess you can use cameraNumber = -1

mdisg, 2021-09-10:
@RealSenseSupport no I did not because there is no documentation or information about that behavior.