realsenseai/RealSenseID #128: Add support to d400 cameras

lukacat10, 2021-09-04:
As the f450 face id line of products is discontinued, but stereo d400 series cameras will still be supported, it might be a good thing to support d400 cameras in this library, to avoid losing high quality code and functionality, and to ease the loss of functionality the realsense community is currently having.

Thank you

RealSenseSupport, 2021-09-08:
Hi Lukacat10,

D400 camera can also be used for FA usage based on depth & RGB image output.
https://www.intelrealsense.com/message-to-customers/
D415 or D435 may be alternative product for F455 from the page suggestion.

santicomp2014, 2021-09-14:
Hi, I'm a bit worried about this shift.

In the [website ](https://www.intelrealsense.com/message-to-customers) the F455/F450 will be replaced with D415/D435
`Requires 3rd party facial authentication software to run on host in conjunction with the depth camera.`

Is there any suggestion for such 3rd party software?, or will there be some AI models for face recognition available in D415/D435 models.

Thank you very much.

lukacat10, 2021-10-01:
> Hi, I'm a bit worried about this shift.
> 
> In the [website ](https://www.intelrealsense.com/message-to-customers) the F455/F450 will be replaced with D415/D435
> `Requires 3rd party facial authentication software to run on host in conjunction with the depth camera.`
> 
> Is there any suggestion for such 3rd party software?, or will there be some AI models for face recognition available in D415/D435 models.
> 
> Thank you very much.
> @RealSenseCustomerSupport 

Exactly.

I hope that this library will be the said "3rd party", but this is not a supported option at the moment.

I feel like it's the right thing to do, in order to not waste all the efforts put into this library. But I'm not the decision maker :)