realsenseai/RealSenseID #129: Make hardware open source

JonasJossen, 2021-09-07:
Dear Intel, 

i was very disappointed reading that you guys discontinue LiDAR, Facial Authentication and Tracking product families in Feb 22. In order to still satisfy your giant community of developers and users, it would be a great gesture to (at least) **make these discontinued hardware devices open source**.

I am pretty sure I am not the only disappointed customer that would respect this step.

santicomp2014, 2021-09-14:
I Agree there should be some kind of opensource to this project.
I was wondering if OAK-D could do the trick or use D415/D435, but the AI models should be trained for this exact purpose (Anti-Spoofing, etc).

Not really sure what 3party software with D415/D435 means.
@RealSenseCustomerSupport

luisruizlatic, 2021-12-20:
Hello guys, hope you are doing great! I was wondering if you have a solution to use the camera D415/D435 with this SDK