realsenseai/RealSenseID #135: libusb1.0.so missing and no known rule to make it

Papiryk, 2021-12-11:
Hi,

I am trying to build the Realsense ID app using android wrapper. After many errors and troubles I've finally managed to get successful Gradle sync. But when I try to run AssembleRelease or build or any other Gradle task, I get this error:

FAILURE: Build completed with 2 failures.

1: Task failed with an exception.
-----------
* What went wrong:
Execution failed for task ':externalNativeBuildSecuredRelease'.
> Build command failed.
  Error while executing process C:\Users\Filip\AppData\Local\Android\Sdk\cmake\3.10.2.4988404\bin\ninja.exe with arguments {-C D:\realsense_id_android\RealSenseID-master\wrappers\android\.cxx\cmake\securedRelease\arm64-v8a RealSenseIDSwigJNI cjpeg-static djpeg-static jpegtran-static md5cmp rdjpgcom rsid rsid-cli rsid-fw-update rsid_c rsid_secure_helper wrjpgcom}
  ninja: Entering directory `D:\realsense_id_android\RealSenseID-master\wrappers\android\.cxx\cmake\securedRelease\arm64-v8a'
  
  ninja: error: '../../../../../../3rdparty/uvc/arm64-v8a/libusb1.0.so', needed by 'lib/librsid.so', missing and no known rule to make it

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.
==============================================================================

2: Task failed with an exception.
-----------
* What went wrong:
Execution failed for task ':externalNativeBuildUnsecuredRelease'.
> Build command failed.
  Error while executing process C:\Users\Filip\AppData\Local\Android\Sdk\cmake\3.10.2.4988404\bin\ninja.exe with arguments {-C D:\realsense_id_android\RealSenseID-master\wrappers\android\.cxx\cmake\unsecuredRelease\arm64-v8a RealSenseIDSwigJNI cjpeg-static djpeg-static jpegtran-static md5cmp rdjpgcom rsid rsid_c wrjpgcom}
  ninja: Entering directory `D:\realsense_id_android\RealSenseID-master\wrappers\android\.cxx\cmake\unsecuredRelease\arm64-v8a'
  
  ninja: error: '../../../../../../3rdparty/uvc/arm64-v8a/libusb1.0.so', needed by 'lib/librsid.so', missing and no known rule to make it

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.
==============================================================================

* Get more help at https://help.gradle.org

BUILD FAILED in 1s
12:47:11: Task execution finished 'assembleRelease'.

I tried to search for the solution, but I couldn't find anything, nor I could find anything about libusb1.0.so file. 

I would truly appreciate any help. Thank you.

RealSenseSupport, 2021-12-15:
You can copy the libraries from pre-built android sdk package.
for example, https://github.com/IntelRealSense/RealSenseID/releases/download/v0.25.0/rsid_sdk_v0.25.0_android.zip

Papiryk, 2021-12-16:
@RealSenseSupport It's working! Thank you so much!

closed 2021-12-16 by Papiryk