realsenseai/RealSenseID #143: Update FW failed

lmjieSCU, 2022-04-13:
I tried to update FW from 2.6 to 2.8/3.1.
And I got:
[2022-04-13 21:12:42.458] [debug] [FwUpdater] cannot find DNET : blk 0 sz=524288
[2022-04-13 21:12:42.458] [debug] [FwUpdater] Stopping reader thread..
[2022-04-13 21:12:42.566] [debug] [FwUpdater] Reader thread stopped
[2022-04-13 21:12:42.567] [error] [FwUpdater] Did not receive 'dl ack'
Firmware update failed

lmjieSCU, 2022-04-14:
Thank you for your help

RealSenseSupport, 2022-04-15:
Can you run rsid-viewer before upgrading the FW?
which upgrade tool do you use? I recommend rsid-fw-update v0.25.0

chaiyonga, 2022-06-03:
The problem "update FW from 2.6 to 2.8/3.1." is Firmware update failed.
Why can not update directly from 2.6 to 4.3 ? 
I want to use FW 4.3

closed 2025-04-03 by gabime