realsenseai/RealSenseID #156: authentication error at application startup

rifai-github, 2024-05-27:
a few days ago I bought an Intel RealSense F455 at this link https://store.intelrealsense.com/intel-realsense-id-solution-f455-peripheral-version-with-software.html.

I've tried it but why do I get an error?
this is the error:
[image]

sherrysun2018, 2024-06-12:
@rifai-github Did you register the product key?

RealSenseCustomerSupport, 2024-06-24:
----------------------------------------------

@rifai-github Did you have chance to register the product key? Looking forward to your reply. Thanks!

RealSenseCustomerSupport, 2024-07-03:
----------------------------------------------

@rifai-github Do you still have issue? Could you please update? Thanks!

RealSenseCustomerSupport, 2024-11-04:
----------------------------------------------

@rifai-github  We didn't hear from you for several months. Will close this ticket at this point. If you still have issue, please feel free create another ticket. Thanks!

closed 2024-11-04 by RealSenseCustomerSupport