realsenseai/RealSenseID #157: JPG Image capture/output

viniciuscsf, 2024-06-06:
I wasn't able to capture images from the camera using the sample aplication/dump mode. Is it supported or is there any code available to output a JPG image to the host?

RealSenseCustomerSupport, 2024-06-12:
----------------------------------------------

Hi viniciuscsf,
Would you share the version information about FW and RSID SDK to me?

viniciuscsf, 2024-06-12:
Hi, I'm using the Realsense ID version 0.33.3
Firmware: 5.7.0.301
 * NNLAS - 7.13.21.0
 * RECOG - 8.2.24.0
 * AS2DLR - 13.1.24.0
 * NNLED - 11.2.24.0
 * OPFW - 5.7.0.301
 * DNET - 6.2.24.0
 * YOLO - 26.12.23.0

RealSenseCustomerSupport, 2024-06-19:
----------------------------------------------

Would you try to run the preview sample?
https://github.com/IntelRealSense/RealSenseID/blob/master/samples/cpp/preview.cc

gabime, 2024-07-02:
@viniciuscsf Try running as root [this ](https://github.com/IntelRealSense/RealSenseID/blob/master/scripts/realsenseid_metadata_win10.ps1)script to enable it.

closed 2025-04-03 by gabime