realsenseai/RealSenseID #178: 🔒 SSL Certificate Expired – License Verification Failing

byrash, 2025-04-12:
We're encountering an SSL certificate issue when using the Intel RealSense Vision Platform for license checking. This is blocking our ability to use licensed devices.

📋 Error Details
When trying to use the platform, we're getting a license checker error. Here's what we found in the logs:

```

[2025-04-11 19:03:30.769] [info] [LicenseCheckerImpl] GET license request to https://visionplatform.intelrealsense.com/v2/f450/license-status?license_key=XXXXXXXXXXX&serial_number=XXXXXXXXXXXX&encrypted_session_token=XXXXXXXX
[2025-04-11 19:03:31.043] [info] [LicenseCheckerImpl] HTTP response code -1
[2025-04-11 19:03:31.044] [debug] [LicenseCheckerImpl] License x-request-id: not found.
[2025-04-11 19:03:31.044] [error] [LicenseCheckerImpl] Network issues
[2025-04-11 19:03:31.044] [error] [FaceAuthenticatorImpl] License verification failed
```

Upon deeper investigation, we found the root cause:

```
curl -vI --ssl-no-revoke https://visionplatform.intelrealsense.com 2>&1 | grep "SSL"
* SSL certificate problem: certificate has expired
curl: (60) SSL certificate problem: certificate has expired
```

[image]

⚠️ Impact
License validation fails

Device usage is completely blocked

This is not the first time we've encountered this issue during license activation

Indicates a possible lack of timely SSL certificate renewal on Intel’s side

✅ Request
Please renew the SSL certificate for visionplatform.intelrealsense.com as soon as possible.

Also, we recommend setting up automated certificate renewal or monitoring to prevent future outages like this.

Thanks! 🙏

devloic, 2025-05-16:
@byrash 
did intel give you any alternative to register your license on the device? I modified restclient-cpp so it doesn't check the certificate but the license server returns a HTTP 500 error anyway.

byrash, 2025-05-27:
Hello @gabime , Thanks for that. What does that mean ? For using intel release we no longer need any licenses purchase? Please confirm. Many Thanks